FreeOpcUa/freeopcua #381: Is it unmaintained ?

etiennemabille, 2021-01-12:
The website http://freeopcua.github.io states that this library is not maintained anymore.
If that's the case, I think it should be mentioned clearly in this repository's README

Additionnaly, is there any hope that the maintenance of this library shall be resumed someday ?

pbertoni89, 2021-01-13:
Afaik the most active repo in this org in the Python async API [FreeOpcUa/opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio).

etiennemabille, 2021-01-13:
I understand that all the Python -based projects are still active. 
I meant to refer to the C++ library. This repository is only dedicated to the C++ library, or am I mistaken ?

destogl, 2021-01-13:
Hi @etiennemabille,

the library is more or less unmaintained. I am currently looking to manage at least some pull requests. 

There is  open62541/open62541 implementation which supposedly supports also C++. I am not sure how good C++ support is. But the library is for sure very active.

etiennemabille, 2021-01-14:
Yes I am currently looking at [open62541/open62541](https://github.com/open62541/open62541) and its C++ wrappers ([bazza2jw/open62541Cpp](https://github.com/bazza2jw/open62541Cpp) and [seronet-project/open62541Cpp](https://github.com/seronet-project/open62541Cpp)) 

Thank you very much for your time and answers.

closed 2021-01-14 by etiennemabille