FreeOpcUa/freeopcua #383: Cmake Library error while compiling on Fedora

sunnybunny123, 2021-03-07:
I am facing the following error while compiling on Fedora with cmake. anyone please help

[ 86%] Building CXX object CMakeFiles/example_server.dir/src/examples/example_server.cpp.o
[ 87%] Linking CXX executable bin/example_server
/usr/bin/ld: cannot find -lmbedcrypto
/usr/bin/ld: cannot find -lmbedx509
/usr/bin/ld: cannot find -lmbedtls
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/example_server.dir/build.make:113: bin/example_server]

mbec-lbto, 2021-03-07:
try ```sudo dnf install mbedtls-devel```

sunnybunny123, 2021-03-08:
> try `sudo dnf install mbedtls-devel`

Thankyou very much. It helped that issue.
Now i am trying to run the examples.
Thanks