FreeOpcUa/freeopcua #384: GetChild freeopcua method  requested operation has no match to return. (0x806f0000)

FaidiSaif, 2021-05-23:
Hello Gents, 

I'm trying to use  freeopcua c++ GetChild method to get standard dynamic variable node from the root node of freeopcua-modeler using the following code :      

```
      std::vector<std::string> varpath{ "Objects", "Server", "ServerStatus", "CurrentTime" };
      myvar = root.GetChild(varpath); // problem here.

```
but the i got the _requested operation has no match to return. (0x806f0000) _ error

[image]

have you any idea how to fix this ? 

**below the opcua-modeler server interface :** 

[image]

Note: using the freeopcua [server_example.cpp](https://github.com/FreeOpcUa/freeopcua/blob/master/src/examples/example_server.cpp) as OPC UA server works fine with my client ! but it does not work with this python opcua-modeler interface ! 

**this is my complete freeopcua client code (based on freeopcua example_client.cpp):**
`
#include <opc/ua/client/client.h>
#include <opc/ua/node.h>
#include <opc/ua/subscription.h>

#include <opc/common/logger.h>

#include <iostream>
#include <stdexcept>
#include <thread>

using namespace OpcUa;

class SubClient : public SubscriptionHandler
{
  void DataChange(uint32_t handle, const Node & node, const Variant & val, AttributeId attr) override
  {
    std::cout << "Received DataChange event, value of Node " << node << " is now: "  << val.ToString() << std::endl;
  }
};

int main(int argc, char ** argv)
{
  auto logger = spdlog::stderr_color_mt("client");
  try
    {
      //std::string endpoint = "opc.tcp://127.0.0.1:4840/freeopcua/server/";
      std::string endpoint = "opc.tcp://127.0.0.1:48400/freeopcua/uamodeler/";

      logger->info("Connecting to: {}", endpoint);

      OpcUa::UaClient client(logger);
      client.Connect(endpoint);

      //get Root node on server
      OpcUa::Node root = client.GetRootNode();
      logger->info("Root node is: {}", root);

      //get and browse Objects node
      logger->info("Child of objects node are:");
      Node objects = client.GetObjectsNode();

      for (OpcUa::Node node : objects.GetChildren())
        { logger->info("    {}", node); }

      //get a node from standard namespace using objectId
      logger->info("NamespaceArray is:");
      OpcUa::Node nsnode = client.GetNode(ObjectId::Server_NamespaceArray);
      OpcUa::Variant ns = nsnode.GetValue();

      for (std::string d : ns.As<std::vector<std::string>>())
        { logger->info(" namespace =    {}", d); }

      OpcUa::Node myvar;

      //uint32_t idx = client.GetNamespaceIndex("http://opcfoundation.org/UA/");
   
      std::vector<std::string> varpath{ "Objects", "Server", "ServerStatus", "CurrentTime" };

      myvar = root.GetChild(varpath);

      logger->info("got node: {}", myvar);

      //Subscription
      SubClient sclt;
      Subscription::SharedPtr sub = client.CreateSubscription(100, sclt);
      uint32_t handle = sub->SubscribeDataChange(myvar);
      logger->info("Got sub handle: {}, sleeping 5 seconds", handle);
      std::this_thread::sleep_for(std::chrono::seconds(5));
      logger->info("Disconnecting");
      client.Disconnect();
      logger->flush();
      return 0;
    }
  catch (const std::exception & exc)
    {
      logger->error("Error: {}", exc.what());
    }
  catch (...)
    {
      logger->error("Unknown error.");
    }
  return -1;
}

`

SangKee, 2024-05-23:
I'm having the same problem.
Anybody solve this issue?

guojing555, 2024-05-23:
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。