FreeOpcUa/freeopcua #387: Unable to create value change subscription

apoorvcn47, 2021-07-22:
I am trying the example code giving here
https://github.com/FreeOpcUa/freeopcua/blob/master/src/examples/example_client.cpp

Server in this case is a fanuc robot. My code exits while I try to create value change subscription with following log

[2021-07-22 12:31:26.779] [UaClient] [info] ua_client             | creating session ... 
[2021-07-22 12:31:26.782] [UaClient] [info] ua_client             | create session OK 
[2021-07-22 12:31:26.782] [UaClient] [info] ua_client             | activating session ... 
[2021-07-22 12:31:26.790] [UaClient] [info] ua_client             | activate session OK 
[2021-07-22 12:31:26.791] [UaClient] [info] keep_alive_thread     | starting 
[2021-07-22 12:31:26.804] [UaClient] [info] keep_alive_thread     | stopped 
[2021-07-22 12:31:26.808] [UaClient] [warning] binary_client         | receive ServiceFault from Ser
ver with StatusCode: There is no subscription available for this session. (0x80790000)
[2021-07-22 12:31:26.808] [UaClient] [warning] binary_client         | receive ServiceFault from Ser
ver with StatusCode: There is no subscription available for this session. (0x80790000)
[2021-07-22 12:31:26.812] [UaClient] [info] ua_client             | CloseSession response is  
^C[INFO] [1626971492.387745986] [rclcpp]: signal_handler(signal_value=2)

I know everything with server is fine because I can create value change subscription with python free opcua. Any idea?

zaaczaac, 2021-07-29:
Hello !

I'm a beginner in OPC UA and with that kind of programming.

I'm a bit lost with all that file but I can see you manage to build a project. I'm trying to set up the "example_server.cpp".

I'm using Visual Studio 2019 where can I begin? 

Thanks a lot!

apoorvcn47, 2021-07-29:
I haven't worked with windows in a long time, but all you have to do, is link .h,.so/.lib and .dll. You can do a simple google search to learn how to link these 3 kind of dependencies and its pretty straight forward