FreeOpcUa/freeopcua #388: Some Bugs when building on Windows (uri_facade_win.cpp,socket_channel.cpp AND tcp_server.cpp)

HILMR, 2021-11-03:
1.  Up to now, the master branch still does not merge the modifications described in issue #311 and #312 
2.  There are still errors caused by the different socket interface between Windows and Linux
  - https://github.com/FreeOpcUa/freeopcua/blob/bd13aee2455c1d137bee45f7d2aedea8dd30115c/src/core/socket_channel.cpp#L51-L54
    ```c++
    void OpcUa::SocketChannel::Stop()
    {
      closesocket(Socket); //API For Windows
      //close(Socket); //API For Linux
    }
    ```
- https://github.com/FreeOpcUa/freeopcua/blob/bd13aee2455c1d137bee45f7d2aedea8dd30115c/src/server/tcp_server.cpp#L11-L13
  ``` c++
  #ifdef _WIN32
  #include <WinSock2.h> // "SD_BOTH" declared in this header file
  #include <windows.h>
  #endif
  ```

- https://github.com/FreeOpcUa/freeopcua/blob/bd13aee2455c1d137bee45f7d2aedea8dd30115c/src/server/tcp_server.cpp#L175-L185
  ``` c++
    virtual void Stop()
    {
      if (ServerThread)
        {
          LOG_INFO(Logger, "shutting down opc ua binary server");
          Stopped = true;
          shutdown(Socket, SD_BOTH); //API For Windows
          //shutdown(Socket, SHUT_RDWR); //API For Linux
          ServerThread->Join();
          ServerThread.reset();
        }
    }
  ```

Previously, the python version of this project is very easy to use, but due to some requirements changes, I have to develop the C + + version on windows. However, it is found that the support of this project for windows is not so friendly. I hope the author can further optimize it. Thank you :)

AVerutin, 2022-03-15:
> 1. Up to now, the master branch still does not merge the modifications described in issue [Definition doesn't match declaration in uri_facade_win.cpp/uri_facade.h #311](https://github.com/FreeOpcUa/freeopcua/issues/311) and [Bug fixes for Windows and the OPC UA server #312](https://github.com/FreeOpcUa/freeopcua/pull/312)
> 2. There are still errors caused by the different socket interface between Windows and Linux
> 
> * https://github.com/FreeOpcUa/freeopcua/blob/bd13aee2455c1d137bee45f7d2aedea8dd30115c/src/core/socket_channel.cpp#L51-L54
>   
>   ```c++
>   void OpcUa::SocketChannel::Stop()
>   {
>     closesocket(Socket); //API For Windows
>     //close(Socket); //API For Linux
>   }
>   ```
> * https://github.com/FreeOpcUa/freeopcua/blob/bd13aee2455c1d137bee45f7d2aedea8dd30115c/src/server/tcp_server.cpp#L11-L13
>   
>   ```c++
>   #ifdef _WIN32
>   #include <WinSock2.h> // "SD_BOTH" declared in this header file
>   #include <windows.h>
>   #endif
>   ```
> * https://github.com/FreeOpcUa/freeopcua/blob/bd13aee2455c1d137bee45f7d2aedea8dd30115c/src/server/tcp_server.cpp#L175-L185
>   
>   ```c++
>     virtual void Stop()
>     {
>       if (ServerThread)
>         {
>           LOG_INFO(Logger, "shutting down opc ua binary server");
>           Stopped = true;
>           shutdown(Socket, SD_BOTH); //API For Windows
>           //shutdown(Socket, SHUT_RDWR); //API For Linux
>           ServerThread->Join();
>           ServerThread.reset();
>         }
>     }
>   ```
> 
> Previously, the python version of this project is very easy to use, but due to some requirements changes, I have to develop the C + + version on windows. However, it is found that the support of this project for windows is not so friendly. I hope the author can further optimize it. Thank you :)

Hi. Did you find anything instead this library?

guojing555, 2022-03-15:
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。