FreeOpcUa/freeopcua #389: Terminate called after throwing an instance of 'std::runtime_error'  what(): Response timed out

nbergantino, 2022-04-18:
Connecting a Kepware server after some times the program is aborted with the error in the title. I am using the subscription method.
Observing the frame stack with gdb I detected the following once the exception is thrown:
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7bd8859 in __GI_abort () at abort.c:79
#2  0x00007ffff7e61911 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7e6d38c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7e6d3f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7e6d6fd in __cxa_rethrow () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff2b731bd in (anonymous namespace)::BinaryClient::Send<OpcUa::ReadResponse, OpcUa::ReadRequest> (this=0x7fffe4007720, request=...)
    at /code/cpp/module/service/freeopcua-master/src/client/binary_client.cpp:899
#7  0x00007ffff2b6b1a2 in (anonymous namespace)::BinaryClient::Read (this=0x7fffe4007720, params=...) at /code/cpp/module/service/freeopcua-master/src/client/binary_client.cpp:380
#8  0x00007ffff31df86e in OpcUa::Node::GetAttribute (this=0x7fffe4002248, attr=OpcUa::AttributeId::Value) at /code/cpp/module/service/freeopcua-master/src/core/node.cpp:67
#9  0x00007ffff31e41a0 in OpcUa::Node::GetValue (this=0x7fffe4002248) at /code/cpp/module/service/freeopcua-master/src/core/node.cpp:550
#10 0x00007ffff2bd25ed in OpcUa::KeepAliveThread::Run (this=0x7fffe4002240) at /code/cpp/module/service/freeopcua-master/src/client/client.cpp:84
#11 0x00007ffff2bd20f7 in OpcUa::KeepAliveThread::<lambda()>::operator()(void) const (__closure=0x7fffe4005ed8) at /code/cpp/module/service/freeopcua-master/src/client/client.cpp:46
#12 0x00007ffff2bd6525 in std::__invoke_impl<void, OpcUa::KeepAliveThread::Start(OpcUa::Services::SharedPtr, OpcUa::Node, OpcUa::Duration)::<lambda()> >(std::__invoke_other, OpcUa::KeepAliveThread::<lambda()> &&) (__f=...) at /usr/include/c++/9/bits/invoke.h:60
#13 0x00007ffff2bd64c6 in std::__invoke<OpcUa::KeepAliveThread::Start(OpcUa::Services::SharedPtr, OpcUa::Node, OpcUa::Duration)::<lambda()> >(OpcUa::KeepAliveThread::<lambda()> &&) (
    __fn=...) at /usr/include/c++/9/bits/invoke.h:95
#14 0x00007ffff2bd6464 in std::thread::_Invoker<std::tuple<OpcUa::KeepAliveThread::Start(OpcUa::Services::SharedPtr, OpcUa::Node, OpcUa::Duration)::<lambda()> > >::_M_invoke<0>(std::_Index_tuple<0>) (this=0x7fffe4005ed8) at /usr/include/c++/9/thread:244
#15 0x00007ffff2bd6425 in std::thread::_Invoker<std::tuple<OpcUa::KeepAliveThread::Start(OpcUa::Services::SharedPtr, OpcUa::Node, OpcUa::Duration)::<lambda()> > >::operator()(void) (
    this=0x7fffe4005ed8) at /usr/include/c++/9/thread:251
#16 0x00007ffff2bd63fa in std::thread::_State_impl<std::thread::_Invoker<std::tuple<OpcUa::KeepAliveThread::Start(OpcUa::Services::SharedPtr, OpcUa::Node, OpcUa::Duration)::<lambda()> > > >::_M_run(void) (this=0x7fffe4005ed0) at /usr/include/c++/9/thread:195
#17 0x00007ffff7e99de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#18 0x00007ffff7b39609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#19 0x00007ffff7cd5163 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

guojing555, 2022-04-18:
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

nbergantino, 2022-04-18:
Changing the void KeepAliveThread::Run() as following the issue appears solved:

void KeepAliveThread::Run()
{
  LOG_INFO(Logger, "keep_alive_thread     | starting");

  while (!StopRequest)
    {
		**try{**
		      int64_t t_sleep = Period * 0.7;
		      LOG_DEBUG(Logger, "keep_alive_thread     | sleeping for: {}ms", t_sleep);

		      std::unique_lock<std::mutex> lock(Mutex);
		      std::cv_status status = Condition.wait_for(lock, std::chrono::milliseconds(t_sleep));

		      if (status == std::cv_status::no_timeout)
		        {
		          break;
		        }

		      LOG_DEBUG(Logger, "keep_alive_thread     | renewing secure channel");

		      OpenSecureChannelParameters params;
		      params.ClientProtocolVersion = 0;
		      params.RequestType = SecurityTokenRequestType::Renew;
		      params.SecurityMode = MessageSecurityMode::None;
		      params.ClientNonce = std::vector<uint8_t>(1, 0);
		      params.RequestLifeTime = Period;
		      OpenSecureChannelResponse response = Server->OpenSecureChannel(params);

		      if ((response.ChannelSecurityToken.RevisedLifetime < Period) && (response.ChannelSecurityToken.RevisedLifetime > 0))
		        {
		          Period = response.ChannelSecurityToken.RevisedLifetime;
		        }

		      LOG_DEBUG(Logger, "keep_alive_thread     | read a variable from address space to keep session open");

		      NodeToRead.GetValue();
	    **}catch (const std::exception & exc) {
			printf("KeepAliveThread::Run Rilevato errore %s", exc.what());
		}catch (...)  {
			printf("KeepAliveThread::Run Rilevato errore sconosciuto");
		}**
    }

  Running = false;

  LOG_INFO(Logger, "keep_alive_thread     | stopped");
}