FreeOpcUa/freeopcua #412: UNABLE TO connect to Siemens 828D

IRISBFW, 2024-04-19:
Kindly some one help
I need to connect to siemens 828D Controller through OPCUA
It completely working with the python
Now i need to do the same with C++
I am able to connect to server without username and password
But when i try to connect to the server with username and password, it 

I connect to server with
 std::string endpoint = "opc.tcp://IRIS:12345678@192.168.100.40:4840";

I am below error
receive ServiceFault from Server with StatusCode: The user identity token is not valid. (0x80200000)

guojing555, 2024-04-19:
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

IRISBFW, 2024-05-29:
Can someone help on this?????

IRISBFW, 2024-08-17:
Still awaiting for the replay
Come one help please ???

zhaodadong, 2024-11-01:
std::vector<OpcUa::UserTokenPolicy> GetUserTokens() const
  {
    OpcUa::UserTokenPolicy policy;
    policy.IssuedTokenType = "IssuedTokenType";
    policy.IssuerEndpointUrl = "IssuerEndpointUrl";
    policy.PolicyId = "PolicyId";
    policy.SecurityPolicyUri = "SecurityPolicyUri";
    policy.TokenType = UserTokenType::UserName;
    return std::vector<OpcUa::UserTokenPolicy>(1, policy);
  }
Maybe need confirm some information supplied by the Ua server and add to endpoint?